lncm/pi-factory #5: Make sure pi-shutdown.service, shutdowns the pi

meeDamian, 2018-09-22:


meeDamian, 2018-10-25:
No clue how to fix that :/

Any suggestions welcome

meeDamian, 2018-10-26:
I cannot. Explicit calls do not work, that's there reason I tried going with a service in the first place.

Feel free to PR docker support at any time, stuff should work already, and if something needs correcting we should do that as we go.

The biggest unknown for me at this point is migration to `pi-init3`, if that works everything _should_ 😬

nolim1t, 2018-10-26:
pi-init3 does work.. ive been using it for the last month

meeDamian, 2018-10-26:
Yeah, I've tested it and merged to `master` already too.

But let's keep the issue on the topic of broken shutdown ;P

AnotherDroog, 2018-11-06:
This can be closed since we won't be needing pi-init anymore, right?

meeDamian, 2018-11-07:
This issue is not `pi-init` specific, although we'll be able to get rid of `pi-init` entirely once we migrate to alpine.

meeDamian, 2018-11-08:
Made issue for eventual getting rid of `pi-init3`: #34