lncm/pi-factory #6: cp wpa_supplicant.conf

AnotherDroog, 2018-09-23:
`rm -rf tmp
[ -f wpa_supplicant.private.conf ] && \
		cp wpa_supplicant.private.conf boot/wpa_supplicant.conf || \
		cp tmp/wpa_supplicant.conf boot/wpa_supplicant.conf
cp: tmp/wpa_supplicant.conf: No such file or directory
make: *** [boot/wpa_supplicant.conf] Error 1`

wpa_supplicant.conf exists and is populated with credentials

meeDamian, 2018-09-23:
Yeah, I use a `.gitignored` `wpa_supplicant.private.conf`, because I want `wpa_supplicant.conf` to be a blank template that can be safely committed to a repo.

I'll obv fix it, but I have to figure out how to validate it properly first.

For now you can:

```
cp ./wpa_supplicant.conf ./wpa_supplicant.private.conf
make write_stuff_to_boot SD=/dev/… 
``` 

`SD=` parameter above is optional in case `/Volumes/boot` is not mounted.

Running `make write_stuff_to_boot` skips the lengthy image writing process and skips straight to only copying all us-specific files. It's safe to re-run it multiple times **as long as the card wasn't run in the RBP yet**.

closed 2018-09-23 by meeDamian

AnotherDroog, 2018-09-23:
Very good to know about ‘make write_stuff...’

commit c988b503 (2018-09-23): fix wpa_supplicant handling; fixes #6