lncm/pi-factory #7: Shutdown RBP on missing `block/` and/or `chainstate/`

meeDamian, 2018-09-23:
Also figure out how to do `rsync` / get them in a way that's not a PITA.

meeDamian, 2018-10-25:
Guys, how should we do the chain bootstrapping? 

How much extra resources would it take to have a 2nd weekly snapshot of minimally pruned chain and an IPFS/bittorrent instance running on each Pi?

Worth it? Any good alternative suggestions?

meeDamian, 2018-10-25:
I really like the Keybase idea!

We can also put there our individual signatures of the downloadable BLOB!

That seems to be the easiest, while still good solution. Once that's done we can experiment with other, more effort solutions

nolim1t, 2018-10-25:
Only thing with keybase is that people can only download the files if you're part of the team - which adds some friction to it.

meeDamian, 2018-10-25:
Public files too?

nolim1t, 2018-10-25:
There’s no concept of public and private with teams.

meeDamian, 2018-10-26:
Then perhaps we should host it on our public accounts. It **has to** be accessible without any authentication as it's meant to happen in a fully automated fashion on the pi

meeDamian, 2018-11-06:
Research how to do it on Alpine

AnotherDroog, 2018-11-06:
We’ve got `poweroff` on Alpine. Obviously a container will need special privileges to execute this. 

Alternatively it can be a call on the local api we are offering for setup already

AnotherDroog, 2018-11-06:
Isn’t it enough to halt the container? Why shutdown?

AnotherDroog, 2018-11-06:
@nolim1t Docker containers can in fact execute commands on the host, they must be run in privileged mode to do so.

meeDamian, 2018-11-06:
You're right, it doesn't make sense in our current setup - it should be communicated to user in some other way, therefore closing this issue.

closed 2018-11-06 by meeDamian