lncm/pi-factory #8: Add docker

AnotherDroog, 2018-09-25:
`curl -fsSL https://get.docker.com -o get-docker.sh`
`sudo sh get-docker.sh`

nolim1t, 2018-10-10:
that would be cool to have but also having problems trying to build a docker image on the PI zero for some reason so will have to tweak that dockerfile

nolim1t, 2018-10-25:
I have something working which I'll add once @meeDamian completes #12

meeDamian, 2018-10-26:
I'd say we install docker `curl -fsSL get.docker.com | sh` on `box` and `blank`, but pull & setup all images only on the `box`.

Also, there's a `pi-setup.sh` file already on master with variants and `pi-init3` merged-in so if you want to go ahead wit it @nolim1t, feel free ^^.

nolim1t, 2018-10-26:
See pull request #19

meeDamian, 2018-10-27:
I left some comments there, let's discuss modules and some standardisation of docker images?

AnotherDroog, 2018-10-27:
Sure, I’ll gather my thoughts tomorrow

meeDamian, 2018-10-29:
Docker installation added to the `builder` variant. More to follow.

AnotherDroog, 2018-11-03:
Docker has also been added to variant-alpine, including steps necessary to install docker-compose and deps.

meeDamian, 2018-11-06:
I think we should focus exclusively on porting everything to `variant-alpine`, make it default, and gradually rewrite other variants to also be alpine-based.

Does it warrant another issue?

AnotherDroog, 2018-11-06:
Since we’ve got Docker working on Alpine, this issue can be closed and we can open new ones related to porting on a case by case basis

closed 2018-11-06 by meeDamian