lncm/pi-factory #16: Allow for binary installation

meeDamian, 2018-10-26:
I think the best way to go about it is to add `-source` suffix detection to `bitcoind-version`.

As in `0.17.0` installs a binary, and `0.17.0-source` installs it from source.

Some open questions:

* build SD card can be insert to any RBP, how to detect what architecture of the binary to download? cc @AnotherDroog 
* do we still need that option if we use docker? cc. @nolim1t 
* should that version be used to choose docker image by tag?
* should we also consider adding a `-nowallet` (or, depending on what's our default, `-wallet`) tag? ex. `0.17.0-source-nowallet`

nolim1t, 2018-10-26:
I have a binary download tool which downloads the appropriate binary

nolim1t, 2018-10-26:
the docker image has a pre-built binary that we compiled. if theres no trust the person building can also build their own docker image.

meeDamian, 2018-10-26:
I mean the image _is_ a binary blob, I see no reason why we can't just build it from source every time we release a new, tagged image. It's literally a one-time (per bitcoind update) thing.

meeDamian, 2018-11-06:
Obsolete:

1. Either install from source, or
1. Use a docker container

Closing for now, as I won't be spending time on that, if somebody wants to take it, feel free to reopen

closed 2018-11-06 by meeDamian