lncm/pi-factory #25: Convert Makefile to Go

meeDamian, 2018-10-30:
While the Makefile was the right initial choice, the project has evolved enough to justify re-writing it to a language where writing more-complex logic is not pure insanity.

I think Go might be the best choice, mainly because it produces self-contained binaries that users can just double-click to get things done.

I'll be porting the logic gradually, and pushing changes to: https://github.com/lncm/pi-factory/tree/go 

Obv feel free to contribute y'all ^^

nolim1t, 2018-10-30:
ACK if we can build binaries.

AnotherDroog, 2018-11-02:
Python might be a good choice too since it enables us to add a GUI easily (e.g. Qt)

It also allows for self-contained binaries, on multiple platforms.

Or maybe we can combine a golang cli tool with a Python GUI somehow

AnotherDroog, 2018-11-29:
We can close this in light of the alpine architecture

I'm going to go over your Makefile and reuse it for runonce.sh

nolim1t, 2018-11-29:
I'm going to create a "runonce.sh" equivalent which works for alpine... watch the pull requests over the next few days ;p