lncm/pi-factory #30: Install variant-alpine to disk

AnotherDroog, 2018-11-02:


AnotherDroog, 2018-11-19:
With the v0.2.1 release we've settled on a 256MB primary FAT32L partition.

The rest of the SD card can therefore be partitioned on first boot to use `ext2` which has higher performance and preserves the life-span of the card, or `ext3/4` with journaling which amplifies writes but is more resilient and won't need fsck.

Comments @meeDamian ?

meeDamian, 2018-11-19:
256 sounds like a good choice - should fit any SD card, but also be able to store all initial stuff we might want to put there. 

As for the extX system, I'd probably just go with ext4, because why not, and less things to remember :P

AnotherDroog, 2018-11-19:
I can't confirm it myself but apparently Android happily uses `ext4` too, so the impact on wear levelling can't be _that_ bad. Also nice we don't have to worry about fsck at boot.

It's settled then 👍

AnotherDroog, 2018-12-10:
Done

closed 2018-12-10 by AnotherDroog