lncm/pi-factory #31: Add cache dir to variant-alpine

AnotherDroog, 2018-11-02:


AnotherDroog, 2018-11-03:
@meeDamian We need to decide if we want to ship Docker in the initial image (+40MB) or download it once internet connectivity is established.

This determines if we use the iotwifi container as is, or use the golang binary standalone, in concert with dnsmasq & hostapd.

meeDamian, 2018-11-03:
I'd say ship it with the SD card (or burn it onto the card upon setup) but not bundle it in the repo.

The thing to consider is that w/o it the shipped docker will likely be outdated and we should upgrade once the internet is acquired.

AnotherDroog, 2018-11-03:
Since it's working nicely already, I'm also leaning towards keeping iotwifi as a docker container.

Providing Docker alongside the apkovl.tar.gz (or inside) allows users to manually do the steps the installer takes easily. That is desirable in terms of user freedom and ease of auditing.

Technically speaking, we can either put it inside the apkovl (which is pushing what they were designed for) or compress the apk cache dir and ship that. This makes adding extra packages an optional choice, since the system can boot fine without the apk cache.

If we don't want to place cache inside the repo, we could link to the official Docker image in Alpine. Unfortunately, this adds multiple steps to manual installation since Docker, avahi and all dependent apks will have to be downloaded and copied manually.

meeDamian, 2018-11-03:
What's `apkovl.tar.gz`? Also could you rephrase that paragraph, I'm not sure I've understood it.

Why would these binaries need to be downloaded manually?

I think our installer script, upon being run, should download all necessary things:
* newest alpine image, 
* all apk's that we'll use,
* `iotwifi` docker image (although I'm not sure how easy it is to replicate `docker pull cjimti/iotwifi && docker save -o iotwifi.docker cjimti/iotwifi` without the requirement to have Docker installed)

AnotherDroog, 2018-11-03:
What I'm trying to say is, the installation of alpine is so simple that we don't need to force people to use an installer at all. It's nice to have for inexperienced users but many power users will be fine extracting a tar.gz and copying our apkovl.

apkovl (alpine package overlay) is their method of modifying a base system at boot. Similarly to how docker uses overlay filesystems in its layers, alpine can inject modifications to the running filesystem tree.

Therefore, instead of making the installer do a lot of things (and having to port everything to Mac, Linux, Windows), I think it will be better to minimise what it needs to do. Ideally, the raspberry handles as much as possible–while minimising resource usage (which alpine is very good at).

For instance, currently the iotwifi docker image is already being shipped (on my test SD) and doesn't have to be installed by docker or the installer. This saves precious resources while installing and during first-boot.

The way apks work, they don't get installed. Instead, they simply get "added" to the system. You can think of the apk cache as being symlinked into the appropriate places in the filesystem tree.

meeDamian, 2018-11-03:
Well, I want to have an automated way to get an SD card set-up from nothing to everything.

I agree that the initial setup should contain as little stuff as possible, and that alpine is perfectly fine managing itself.

The role of the "installer" or "card writer" or "pi-factory", the way I see it would be:
* downloading fresh versions of all necessary files (from their original sources, so we don't have to duplicate them into our repo)
* ensuring that all pieces fit together nicely (as in: no incompatible versions are used, which can happen if users do things manually)
* the `pi-factory` should probably function as a backend that can be used either via CLI or GUI of some sorts, and for that it's best if it's configured via JSON passed either: as a file reference, as raw json string, or as a set of flags - dunno about specifics here
* writes everything to a card so that no external tools are needed and users don't need to drag'n'drop stuff
* ideally, already partitions the SD card into immutable OS + mutable data sections (if easy/doable via Go standard lib stuff); otherwise alpine can do it to itself later(?)

I'll probably be also adjusting these thing as-I-write the code.

About bundling the `iotwifi` already into the card upon the first boot - it seems that DL-ing Docker images is not as easy as it should be (docker binaries do that through private, undocumented APIs 😞 - [this] is the closest I've found so far), so after all it might be better to ship that wifi-binary in a _"naked"_ state…

[this]: https://devops.stackexchange.com/questions/2731/downloading-docker-images-from-docker-hub-without-using-docker

meeDamian, 2018-11-03:
And, you're right, we don't need to force them to use it at all, we can have (at least the basic) steps to get it up and running described somewhere in `README.md`.

AnotherDroog, 2018-11-03:
Agree with all of this, two observations:

Docker apk exists and is written and maintained by alpine lead developer. Therefore, we don't need to download and install Docker using the sh script, which by the way has several issues pointed out by Docker themselves, namely: 

"Docker provides convenience scripts at get.docker.com and test.docker.com for installing **edge** and **testing** versions of Docker CE into *development* environments quickly and non-interactively. The source code for the scripts is in the docker-install repository. **Using these scripts is not recommended for production environments**, and you should understand the potential risks before you use them"

We can talk about the details of partitioning and what data goes on which storage device in #30

AnotherDroog, 2018-11-03:
`docker save` looks useful, I'll try that out to make the shipped tarball cleaner!

AnotherDroog, 2018-11-04:
There is a satisfactory solution to the chicken/egg problem, we fetch docker.apk from alpine mirrors and put it inside the cache directory.

This can be done by the installer, and equally easily manually.

By using `docker save` we can ship iotwifi cleanly from our repo.

Since the software inside iotwifi is mature and unlikely to change this is not a problem.

Generating the iotwifi docker image using `save` can be done in CI in an auditable way (and equally easily manually using the Dockerfile)

This way we solve both cases in their native vendor-supported ways. 

In the unlikely event that an outdated image or apk gets booted from SD, our script can update it when internet is available.

meeDamian, 2018-11-04:
My only concern is, of course, if it's **possible** to verify that the shipped `docker save`d thing is actually what we claim it is. If it is, and there's no other way to fetch it client-side during pre-setup, then wondering it into our repo might make sense, if it is not possible to verify, then I'm obviously against that approach :). Talk more in person soon!

AnotherDroog, 2018-11-10:
Done using `docker save`. This creates a tar file of containers, which is easily auditable and replaceable.

closed 2018-11-10 by AnotherDroog

meeDamian, 2018-11-11:
Quick note on `docker save` auditability: While images are not deterministic upon build, if we `docker pull` and `docker save` a specific version on two different machines the output files are bitwise identical, so even if we bundle this file in/download it from somewhere else, it's still possible to compare checksums and ensure that files are exactly identical.