lncm/pi-factory #42: runonce.sh

AnotherDroog, 2018-11-29:
This script will initiate persistent installation on first boot

What needs to be done on first boot and only once?

 If "/etc/installed" does not exist execute runonce.sh and:
1. Partition SD card
1. Touch `installed` file in `/etc` to indicate runonce.sh is complete
1. `lbu commit`

Once SD card is partitioned we can pull [portainer](
https://www.portainer.io)/ and more containers (bitcoind, lnd), etc

nolim1t, 2018-11-29:
Was actually doing something similar but only happens after the box gets internet access for the first time, but I guess I can also do it without the net dependency too.

Thoughts?

nolim1t, 2018-11-29:
Thats basically what im doing. 

Execute a script from a file placed in the SD card, then clean up.

Should we do it on first boot or after first internet access or both.

AnotherDroog, 2018-11-30:
I would prefer this being an init script so we can use the dependency system, check for networking, etc

Also simplifies disabling the script after installation using rc-update

AnotherDroog, 2018-12-04:
I’m thinking we put system relevant things into the lncm init script and docker related ones in a separate one

AnotherDroog, 2018-12-04:
E.g. swap and mount points still go into lncm script

nolim1t, 2018-12-04:
Ok I’ll add a swap img file as the next step and see how it goes, then if all goes well will mount the docker folder

AnotherDroog, 2018-12-10:
Added swap creating and installing

AnotherDroog, 2018-12-10:
Next steps are:
- add more usb storage devices
- use multiple storage devices as swap "RAID", prioritise according to wear levelling
- add more containers; lnd, bitcoind, invocing-ui

AnotherDroog, 2018-12-12:
Splitting into separate issues

closed 2018-12-12 by AnotherDroog