lncm/pi-factory #47: Handle usb storage devices

AnotherDroog, 2018-12-12:
- Format and mount storage devices
- Create and add swap with slightly higher priority

nolim1t, 2018-12-16:
I’d leave formatting out for mass storage.

A particular use case would be syncing bitcoind on a desktop or faster machine and then plugging the drive in.

AnotherDroog, 2018-12-28:
Started doing this in `ash`, the limitations (no arrays, etc) suggest I better rewrite in Python, which is conveniently installed at this point by docker-compose anyways

AnotherDroog, 2018-12-28:
As discussed previously there are 3 USB storage devices in our default, supported configuration:
1. **"IMPORTANT"** – _critical files such as node configs, keys_ (could be less than 16GB)
2. **"ARCHIVE"** – _recoverable files such as blocks + chainstate_ (at least 16GB)
3. **"VOLATILE"** – _write intensive files such as swap, logs_ (suggested size 16GB)

AnotherDroog, 2019-01-04:
Solved symlink issues by directly bind-mounting from docker-compose into usb device paths

AnotherDroog, 2019-01-05:
Completed using python3 `lncm-usb.py` which is called by `lncm-post` and in turn calls `lncm-usb` with 4 positional arguments, three partition names sorted by size and the size of the largest in bytes.

closed 2019-01-05 by AnotherDroog