lncm/pi-factory #48: Add lnd container

AnotherDroog, 2018-12-12:
- Handle configuration
- Use usb storage volume or fallback to SD if enough space

nolim1t, 2018-12-13:
configuration is handled in #56

nolim1t, 2018-12-18:
this works on zero and raspbian. A few things need to be tailored here.

```bash
#!/bin/bash

if command -v go 2>&1 1>/dev/null; [ "$?" -ne "0" ]; then
    wget https://dl.google.com/go/go1.11.4.linux-armv6l.tar.gz
    sudo tar xvfz go1.11.4.linux-armv6l.tar.gz -C /usr/local
    export PATH=$PATH:/usr/local/go/bin
else
    export PATH=$PATH:/usr/local/go/bin
    echo "Go already installed"
fi

mkdir -p $HOME/gocode

if [ -z "$GOPATH" ]; then
    echo "setting up gopath"
    export GOPATH=$HOME/gocode
    export PATH=$PATH:/usr/local/go/bin:$GOPATH/bin
    echo "export GOPATH=$HOME/gocode" >> $HOME/.bashrc
    echo "export PATH=$PATH:/usr/local/go/bin:$GOPATH/bin" >> $HOME/.bashrc

    echo "Set up Go!"
else
    echo "gopath already configured"
fi

# Turn off SSL Verification (shrugs.. this is needed for some dependencies which fail)
git config --global http.sslVerify false
go get -d github.com/lightningnetwork/lnd

if ! [ -z "$GOPATH" ]; then
    if ! [ -d "$GOPATH/src/github.com/lightningnetwork/lnd" ]; then
        go get -d github.com/lightningnetwork/lnd
        cd $GOPATH/src/github.com/lightningnetwork/lnd
        echo "Installing LND"
    else
        cd $GOPATH/src/github.com/lightningnetwork/lnd
        make clean
        echo "Upgrading LND"
    fi
    make && make install
else
    echo "GOPATH missing!"
fi

# Turn it back on
git config --global http.sslVerify true
```

AnotherDroog, 2018-12-18:
You can try pulling from `lncm/lnd:latest` now

There's a corresponding `build_lnd.sh` in `dockerfiles/lnd` repo

nolim1t, 2018-12-19:
Sure will check it once I figure out why Issue #67 came up

nolim1t, 2018-12-29:
lnd container is there. The other folders can be mounted on USB if needed

closed 2018-12-29 by nolim1t