lncm/pi-factory #63: ln-cli web

AnotherDroog, 2018-12-16:
Check out and possibly add https://github.com/mably/lncli-web container

AnotherDroog, 2018-12-29:
`docker build . -t lncli-web` from our forked repo currently fails with:

```
> grpc@1.17.0 install /lncli-web/node_modules/grpc
> node-pre-gyp install --fallback-to-build --library=static_library

node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://node-precompiled-binaries.grpc.io/grpc/v1.17.0/node-v57-linux-arm-musl.tar.gz                                                        node-pre-gyp WARN Pre-built binaries not found for grpc@1.17.0 and node@8.14.0 (node-v57 ABI, musl) (falling back to source compile with node-gyp)                                     make: Entering directory '/lncli-web/node_modules/grpc/build'
  CXX(target) Release/obj.target/gpr/deps/grpc/src/core/lib/gpr/cpu_posix.o
  CXX(target) Release/obj.target/gpr/deps/grpc/src/core/lib/gpr/cpu_windows.o
... omitted for brevity...
  CXX(target) Release/obj.target/gpr/deps/grpc/src/core/lib/profiling/basic_timers.o
  CXX(target) Release/obj.target/gpr/deps/grpc/src/core/lib/profiling/stap_timers.o
  AR(target) Release/obj.target/libgpr.a
  COPY Release/libgpr.a
  CXX(target) Release/obj.target/grpc_node/ext/byte_buffer.o
  CXX(target) Release/obj.target/grpc_node/ext/call.o
In file included from ../ext/call.cc:26:0:
../ext/call.h:108:8: error: 'AsyncResource' in namespace 'Nan' does not name a type
   Nan::AsyncResource *async_resource;
        ^~~~~~~~~~~~~
In file included from ../ext/call.cc:27:0:
../ext/call_credentials.h:82:8: error: 'AsyncResource' in namespace 'Nan' does not name a type
In file included from ../ext/call.cc:27:0:                                                                                                                                   
../ext/call_credentials.h:82:8: error: 'AsyncResource' in namespace 'Nan' does not name a type
   Nan::AsyncResource *async_resource;
        ^~~~~~~~~~~~~
../ext/call_credentials.h: In constructor 'grpc::node::plugin_callback_data::plugin_callback_data(const char*, grpc_credentials_plugin_metadata_cb, void*)':
../ext/call_credentials.h:71:9: error: class 'grpc::node::plugin_callback_data' does not have any field named 'async_resource'                                                                  async_resource(NULL) {                                                                                                                                                                 ^~~~~~~~~~~~~~
../ext/call_credentials.h:73:5: error: 'async_resource' was not declared in this scope
     async_resource = new Nan::AsyncResource("grpc:plugin_callback_data");
     ^~~~~~~~~~~~~~
../ext/call_credentials.h:73:26: error: expected type-specifier
     async_resource = new Nan::AsyncResource("grpc:plugin_callback_data");
                          ^~~                                                                                                                                                          ../ext/call_credentials.h: In destructor 'grpc::node::plugin_callback_data::~plugin_callback_data()':                                                                                  ../ext/call_credentials.h:76:12: error: 'async_resource' was not declared in this scope
     delete async_resource;
            ^~~~~~~~~~~~~~
../ext/call.cc: In constructor 'grpc::node::tag::tag(Nan::Callback*, grpc::node::OpVec*, grpc::node::Call*, v8::Local<v8::Value>)':
../ext/call.cc:450:7: error: class 'grpc::node::tag' does not have any field named 'async_resource'
       async_resource(NULL),
       ^~~~~~~~~~~~~~
../ext/call.cc:454:3: error: 'async_resource' was not declared in this scope
   async_resource = new Nan::AsyncResource("grpc:tag");  // Needs handle scope.
   ^~~~~~~~~~~~~~
../ext/call.cc:454:24: error: expected type-specifier
   async_resource = new Nan::AsyncResource("grpc:tag");  // Needs handle scope.
                        ^~~                                                                                                                                                            ../ext/call.cc: In destructor 'grpc::node::tag::~tag()':                                                                                                                               ../ext/call.cc:460:10: error: 'async_resource' was not declared in this scope
   delete async_resource;
          ^~~~~~~~~~~~~~
../ext/call.cc: In function 'void grpc::node::CompleteTag(void*, const char*)':
../ext/call.cc:476:41: error: 'struct grpc::node::tag' has no member named 'async_resource'
     callback->Call(2, argv, tag_struct->async_resource);
                                         ^~~~~~~~~~~~~~
../ext/call.cc:479:41: error: 'struct grpc::node::tag' has no member named 'async_resource'
     callback->Call(1, argv, tag_struct->async_resource);
                                         ^~~~~~~~~~~~~~
make: *** [grpc_node.target.mk:176: Release/obj.target/grpc_node/ext/call.o] Error 1
make: Leaving directory '/lncli-web/node_modules/grpc/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)                                                                [0/1918]
gyp ERR! stack     at emitTwo (events.js:126:13)
[... cut ...]
npm ERR! grpc@1.17.0 install: `node-pre-gyp install --fallback-to-build --library=static_library`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the grpc@1.17.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-12-29T10_07_05_803Z-debug.log
The command '/bin/sh -c apk add --no-cache nodejs npm python make g++ && npm install &&  ./node_modules/.bin/gulp bundles' returned a non-zero code: 1
```

AnotherDroog, 2019-01-23:
Check out this one: https://github.com/djmelik/lndash

AnotherDroog, 2019-04-25:
Would be great to get lndash built on ARM

nolim1t, 2019-10-06:
Closing in favor of #236  we can add this into a separate module

closed 2019-10-06 by nolim1t