lncm/pi-factory #74: box fails to boot up after persistent stage

nolim1t, 2018-12-27:
Have tried on two different RPI boxes 

.. part of the process I see that /media/sd/boot does not get created with setup-disk

I think its a RPI3

nolim1t, 2018-12-27:
[image]

Not being able to copy over the boot partition might affect this?

AnotherDroog, 2018-12-28:
I see what happens here, key line is `ERROR: linux-rpi2...` not found. This is the kernel that's missing!

Possible fix would be to `apk update` to ensure there is no mismatch between cached package index and mirror.

I'll make the necessary change to the installation scripts

Not sure why I can't replicate the issue tho...