lncm/pi-factory #92: Make sure tor is working as intended

AnotherDroog, 2019-01-06:


nolim1t, 2019-01-07:
What’s the current result and expected?

AnotherDroog, 2019-01-07:
There’s a remaining issue with tor uri not being written to lnd.conf if an existing config is detected

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nolim1t, 2019-01-09:
it probably isn't because the script relies on certain items to be present in the lnd.conf file to be written in.

I currently look for the presence of:

```
; externalip=
```
and then replace it with the tor uri (and uncomment it)

lnd is also a bit picky on the positioning of where certain items are as well so we can't just append it.

comments/suggestions for this problem is open

AnotherDroog, 2019-01-11:
I'll come up with some python to edit the lnd file safely

PR #106 (2019-01-15): Backup & replace existing bitcoind, lnd configs
This ensures prune=% and externalip get set correctly

Also closes #92
closed 2019-01-15 by AnotherDroog