lncm/pi-factory #105: lnd filters old blocks

AnotherDroog, 2019-01-14:
After unlocking lnd (reproducibly) prints lots of these messages:

```
LNWL: Filtering block 555853 (00000000000000000023fd370ca75e76b803e90776ff87435161960149ce4a0a) with 2476 transactions
```

Always starts at the same height, up to current block height

AnotherDroog, 2019-01-14:
Beginning looks like this
```
2019-01-14 07:41:30.351 [INF] LTND: Waiting for chain backend to finish sync, start_height=558467
2019-01-14 07:41:34.083 [INF] LNWL: Done catching up block hashes
2019-01-14 07:41:34.189 [INF] LNWL: Started rescan from block 00000000000000000029bf4c06ca7d84de5df484b1a2fba29491c34c3ddb7ac2 (height 555610) for 0 addresses
2019-01-14 07:41:34.189 [INF] LNWL: Starting rescan from block 00000000000000000029bf4c06ca7d84de5df484b1a2fba29491c34c3ddb7ac2
```

nolim1t, 2019-01-14:
Wouldn’t that be an lnd issue?

AnotherDroog, 2019-01-14:
I'm thinking it's related to docker volume mounts in some way.

It would be helpful if you could try to reproduce the issue

AnotherDroog, 2019-01-14:
I can confirm `wallet.db`, `channel.db` have been written to recently

AnotherDroog, 2019-01-15:
Logs during shutdown `lncli stop`:
```
2019-01-15 10:55:04.160 [INF] PEER: unable to read message from 184.58.164.106:9735: read tcp 172.16.88.88:33766->184.58.164.106:9735: use of closed network connection
2019-01-15 10:55:04.161 [INF] SRVR: Disconnecting from 51.68.250.244:9735
2019-01-15 10:55:04.162 [DBG] SRVR: removing peer 51.68.250.244:9735
2019-01-15 10:55:04.162 [INF] PEER: Disconnecting 51.68.250.244:9735, reason: server: disconnecting peer 51.68.250.244:9735
2019-01-15 10:55:04.163 [INF] PEER: unable to read message from 51.68.250.244:9735: read tcp 172.16.88.88:51394->51.68.250.244:9735: use of closed network connection
2019-01-15 10:55:04.254 [DBG] SRVR: Peer 46.101.112.24:9735 has been disconnected
2019-01-15 10:55:04.258 [DBG] SRVR: Peer 13.229.234.94:9735 has been disconnected
2019-01-15 10:55:04.261 [DBG] SRVR: Peer 184.58.164.106:9735 has been disconnected
2019-01-15 10:55:04.265 [DBG] SRVR: Peer 51.68.250.244:9735 has been disconnected
2019-01-15 10:55:04.265 [INF] RPCS: Stopping RPC Server
```

AnotherDroog, 2019-01-16:
Turns out this is expected behaviour when there's no UTXOs!

https://github.com/lightningnetwork/lnd/issues/2326#issuecomment-447132464

closed 2019-01-16 by AnotherDroog