lncm/pi-factory #126: Integrate FastSync to speed up initial chain sync

AnotherDroog, 2019-01-23:
https://github.com/btcpayserver/btcpayserver-docker/tree/master/contrib/FastSync

nolim1t, 2019-01-24:
can this be used to sync the entire blockchain? 

Lets say if you decide not to go with pruned?

AnotherDroog, 2019-02-06:
This will be for pruned, about 1000 blocks I would say

nolim1t, 2019-02-11:
Ideally we make use of [IPFS](https://gateway.ipfs.io/ipfs/QmPJu3J8pY8iRMUnzTJaofX3toKrY3AjoCz1PoBNd2DetQ) to facilitate. I've added an issue for this #159

meeDamian, 2019-02-11:
I wonder if:

1. We start from the same copy of the blockchain, 
1. Run it on multiple bitcoin nodes,
1. Stop them all after exactly the same block

Will the contents of `blocks/` and `chainstate/` be exactly the same? 🤔 

Would allow for at least some level of auditability.

That way we could, ex:

* Host full, archival chain until block X,
* Host pruned recent chunks

That way anyone willing to audit our "fast-sync" chunk would start from our archival part, setup bitcoind to the same level of pruning, stop bitcoind after a specified block and compared their own state with the chunk hosted by us.

Gotta investigate that.

AnotherDroog, 2019-03-18:
Added fastsync to `noma fastsync`

closed 2019-03-18 by AnotherDroog