lncm/pi-factory #128: Standardize "apps" interfaces

meeDamian, 2019-01-28:
Make sure they're:

* **familiar** - container doesn't change conventional port used by the app within
* **versatile** - all app configuration is available to the container as well
* **composable** - all defaults allow for easy app composition - ex. it's enough to just install bitcoind & lnd to have them work together (might require some extra scripting)

Random thought on _composable_ & RPC credentials, perhaps we could generate RPC credentials based on the ID of the container using them - that way we can easily add **and remove** creds from the config file.

nolim1t, 2019-01-28:
All  services work by default.

We only change the ports if there is a conflict.

However with web services, everything will run on port 80 through nginx.

meeDamian, 2019-01-29:
I was talking about the architectural design of individual containers.

I don't want unnecessary entrypoints, rewriting paths, making some things inaccessible/unfhangeable from outside.

We should also come up with a consistent tags for all images built by us, and we should also build all `amd64` images in CI by default.

If we do that, we can be even (hopefully temporarily) uploading `arm` manually and putting a docker manifest in front of them via Travis.

Happy to set it up.

nolim1t, 2019-01-29:
amd64 doesn’t really help find build issues. 

What works on amd64 doesn’t give much guarantees that it is going to run on the rpi hardware.

That’s the challenge right now.

nolim1t, 2019-01-29:
The good thing is what works on rpi hardware (arm) will work on desktops

meeDamian, 2019-01-29:
But having our stack runnable on normal computers is not much work, but opens many more doors into how our software can be used.

nolim1t, 2019-10-07:
Noma addresses a lot of these issues which can be installed on top of this (or a desktop or VPS)

closed 2019-10-07 by nolim1t