lncm/pi-factory #129: Document architecture of The Box

meeDamian, 2019-01-28:


meeDamian, 2019-01-31:
Going through the stuff that has been built so far.

I don't want to open a 1000 new issues, so let's have a discussion about stuff here, and open issues if we deem the change worthy.

meeDamian, 2019-01-31:
So the first thing `GENERATEDRPCAUTH` within `/home/lncm/bitcoin/bitcoin.conf`, it's a good idea to have it if only one set of credentials will ever be added to `bitcoin.conf`, but if we're aiming at modularity, we should probably have a bit smarter script that generates pairs of credentials per each app requesting access.

The first idea that came to my mind is generating credentials based off container ID of the app requesting access to bitcoind. 

The benefits are:
* it's easy to verify if credentials are already added
* it's easy to know which credentials are no longer necessary and remove them

Downside right now would be:
* app upgrade (container ID changes) would necessitate credentials change

meeDamian, 2019-01-31:
## `docker-compose.yml`

* Why are we using `version: '2.1'`, if the newest is `3.7`?
* If we map port `X` to `X`, it's enough to use `"X"` syntax instead of `"X:X"`.
* How are IP addresses of containers chosen?
   * Should we have some convention for it? 		 
   * Should we have a wiki with reference table for them?
* How are container names chosen? Can we use just plain names of apps within instead?
* Should we have a convention for how our Docker images are tagged? (see below)
* Why is `bitcoin.conf` mounted in the container both in `/root/` and `/home/bitcoin/`? Don't we know where it's needed?
* I love pendrive mountpoints! `/home/{important,volatile,archive}/` makes it super easy to understand 😀!
* Should we exposed have ports documented?
* Is it possible to split compose file into multiple files? Would make it nicer for us to do composability.

### Docker image naming

To standardise naming, I recommend following the following pattern:

Assuming app `lnd` at version `v0.5.2`.

#### GH repo

```
github.com/lncm/docker-lnd
``` 
With GH tags corresponding to versions.  I recommend only following stable releases.

Alternatively all `Dockerfile`s in one repo, but having them in separate ones would make setting up automated build pipelines easier (trigger upon version tag).  But it's probably also doable if all are within the same repo.

#### Docker

After exploring a few projects, it seems to me that the convention of doing multi-arch builds now is:
* individual and arch-specific containers are not tagged (not yet sure how to push an untagged image yet…),
* each of the built containers is annotated with proper OS and arch, ex: `docker manifest annotate lncm/MANIFEST_NAME:VERSION BUILT_IMAGE_ID1 --os=linux --arch=arm --variant=v6`
* after all architectures are built an aggregate manifest is created: `docker manifest create lncm/MANIFEST_NAME:latest BUILT_IMAGE_ID1 BUILT_IMAGE_ID2`
* and pushed: `docker manifest push lncm/MANIFEST_NAME:latest`

After that we end up with something like:

```
lncm/lnd:v0.5.1
lncm/lnd:v0.5.2
lncm/lnd:latest
```

Where executing `docker pull lncm/lnd:latest` pulls the latest available version in the right architecture!

Bonus question, should we use `v` in the version string, or skip it?

meeDamian, 2019-01-31:
I see a few files (ex. `lightning/config`, `litecoin/litecoin.conf`) that are not used, and should probably never be part of the default setup. I think it makes sense to get rid of them from the initial repo, and only pull them as part of the "app install process", which AFAIU is yet to be designed?

meeDamian, 2019-01-31:
Note to self: add `README.md` file links to repos/homepages of all projects being used by us.

meeDamian, 2019-01-31:
What for and why exactly is nginx used right now?

meeDamian, 2019-01-31:
Let's not use `.gitkeep` files, they are not recognised by git in any way, and it's better to use conventional `.gitignore` in their place.

meeDamian, 2019-01-31:
IDEA: Each installed app gains access to `/home/lncm/APP_NAME/` namespace where it can put all initial config files it needs to run.

meeDamian, 2019-01-31:
Second `README.md` documenting stuff in `/usr/local/sbin/` added here: https://github.com/lncm/pi-factory/commit/9e1e399f92187d1a0b4431709cb73e604fcd3dfb ([`cleanup`] branch).

[`cleanup`]: https://github.com/lncm/pi-factory/commits/cleanup

meeDamian, 2019-02-01:
> we are mounting as within the container because we can easily invoke commands within the container without specifying a pathfile

I don't know what you're replying to.

I wrote **a ton** of questions and points that should be addressed above, really not sure what this single sentence refers to :).

nolim1t, 2019-02-01:
Replying to the part about pathnames.. they need to work how they are. For ux reasons too

AnotherDroog, 2019-02-06:
>What for and why exactly is nginx used right now?

@meeDamian it’s used for serving WiFi Manager and can proxy port 80 to other endpoints

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

AnotherDroog, 2019-02-06:
>## `docker-compose.yml` ...

@meeDamian bitcoin.conf directory path appears duplicated to work around bitcoin-cli default path

We can use v in the versions or not, I’m fine with either

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

meeDamian, 2019-02-06:
Or we can just not change the default path from `/root/.bitcoin/` 🤷‍♂️

meeDamian, 2019-02-06:
**Note:** [`lnd` does it], it's simple, consistent and what's the point of having a dedicated bitcoin user within a docker container anyway?

[`lnd` does it]: https://github.com/lightningnetwork/lnd/blob/master/Dockerfile#L21

meeDamian, 2019-02-07:
Rebase on top of `alpine-3.9.0` and merge in there.

meeDamian, 2019-02-15:
Opened issue: https://github.com/lncm/pi-factory/issues/165 for https://github.com/lncm/pi-factory/issues/129#issuecomment-459256724

meeDamian, 2019-02-15:
Killing this issue in favor for a few others.

closed 2019-02-15 by meeDamian