lncm/pi-factory #130: Create an automatic reconnect script

meeDamian, 2019-01-28:


nolim1t, 2019-01-28:
What do you mean by automatic reconnect?

meeDamian, 2019-01-29:
Right.  Since some of the boxes will have odd network setups, and some of them will not be reachable front the outside, and it enough if any side initiates the connection.  They should run their own reconnect to all peers scripts in cron. 

When I say _peers_ I mean _LN peers_, but it might be useful for other software as well (Bitcoin?).

AnotherDroog, 2019-02-06:
Could this be part of `mana`, e.g. `mana reconnect` and then cron runs it periodically.

`mana reconnect` takes a text file from /media/important/nodes.txt and calls `lncli connect` for each hostname

meeDamian, 2019-02-11:
Quick idea:

create gist in a format:

```
{
	"<pubkey>": {
		"domain": "<domain>",
		"ip": "<ip>",
		"port": "<port>"
	},
	…
}
```

That lists all our known nodes.

Then, have a script on the box that fetches the gist periodically, filters it by matching pubkeys, and tries to reconnect with a domain provided there.

Optionally, we can also add a script that runs on our side, that periodically updates IPs in the gist above to ones that are known to be work.

Not very "decentralised", but a quick and easy temporary solution that will already improve the state of things.

AnotherDroog, 2019-03-11:
I've made a simple implementation of this in #191

nolim1t, 2019-03-11:
Could it be done in pure python and use the APIs? 

It's less likely to be error prone and it becomes a more reusable/useful utility?

AnotherDroog, 2019-03-18:
The python implementation can be improved via python docker api, yes

This `noma autoconnect` now only needs to be added to cron automatically