lncm/pi-factory #140: Update Alpine to 3.9

meeDamian, 2019-02-04:


AnotherDroog, 2019-02-05:
Verified, patching modloop won't be necessary after 3.9.1

AnotherDroog, 2019-02-06:
There's box 0.4.1 with alpine 3.9.1 image hosted over IPFS 

https://ipfs.io/ipfs/QmX83AUBMAZ39wXJbNvUkSdEhHkTV9WeLtYyrTobLEoZYg

AnotherDroog, 2019-08-18:
This is done isn’t it? @nolim1t

nolim1t, 2019-08-19:
No we are still on 3.8.2 

We ran into issues with getting wifi fully working.with the setup which is why that branch reminds in limbo

AnotherDroog, 2019-08-19:
Okay, here’s an idea, maybe we can reimplement the Pi Box based on postmarketOS easily. 

Their build tools base on apk templates which themselves contain collections of apks. 

After creating an APKBUILD script (basically lncm/noma/install.sh refactored) noma simply gets added to the list of apks to build an image with.

Pre-built pmOS device templates account for hardware differences (Zero, Pi3, Pi4, etc) and can be extended into templates for release types. 

APKs build on our knowledge with shell and alpine. 

pmOS build tools are conveniently installed on Debian+Alpine systems and written in Python to match Noma and docker-compose.

AnotherDroog, 2019-08-19:
@noaho what’s your take on this?

nolim1t, 2019-08-19:
Sounds good.. do we have a repo or branch for postmarketos implementation yet?

AnotherDroog, 2019-08-19:
Well it would still be called pi-factory wouldn’t it? Or Alpine Factory

nolim1t, 2019-08-20:
If you can work with whats in pi-factory and do a postmarketos implementation of it why then sure.

meeDamian, 2019-08-20:
Perhaps update the name of this issue?

nolim1t, 2019-08-20:
Updated. Because 3.9 is not the latest anymore

nolim1t, 2019-10-06:
Will use latest alpine in #236 and #238

PR #215 (2019-05-12): Update to Alpine 3.9.4
close #140 

- Fixes wifi firmware
- Remove modloop workaround
- Add config.txt
- Reduce gpu_mem; close #79
closed 2019-10-06 by nolim1t