lncm/pi-factory #141: Verify all scripts work with sh

meeDamian, 2019-02-04:
Ideally also set up linter for it.

Let's try to avoid bash?

I recommend `sh` + `python3` for more complex things - opinions?

AnotherDroog, 2019-02-06:
I would recommend the initial stage (`/etc/init.d/lncm`) and related parts to be in `ash`

After the first __reboot__ persistent installation is done.  More complex things can be done with `python3`, which is conveniently installed by `/etc/init.d/lncm-post`

meeDamian, 2019-02-06:
So… it's possible to have `init.d/` scripts written in anything else than they already are?

What's `#!/sbin/openrc-run` ?

AnotherDroog, 2019-02-06:
It’s not really possible, they have to be written in openrc compatible ash. 

However they can hand off control to other callables

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-02-07 by AnotherDroog