lncm/pi-factory #148: zapconnect support

AnotherDroog, 2019-02-06:
As a container and exposing QR conveniently

meeDamian, 2019-02-11:
That should just be a part of the dashboard. We'll probably have integrations with multiple LN wallets, and if there's no standard on how passing connstring/macaroon/tls happens, then we'll need a tab/button for each. 

Creating a "dashboard" label, and these issues should ideally be moved to a "dashboard" repo, once it gets created.

AnotherDroog, 2019-03-07:
go get -d github.com/LN-Zap/lndconnect
cd $GOPATH/src/github.com/LN-Zap/lndconnect
make
`lndconnect` can produce QR codes, or strings

AnotherDroog, 2019-03-22:
Good point! In that case, it can just be a function in `noma`

nolim1t, 2019-03-27:
For reference, source code is at https://github.com/LN-Zap/lndconnect

As long as its in the same format that ZAP expects then its fine

nolim1t, 2019-03-27:
created issue within noma project

AnotherDroog, 2019-04-25:
> created issue within noma project

Thanks! We can also transfer next time :)

closed 2019-04-25 by AnotherDroog