lncm/pi-factory #151: Remove as many files from the repo as possible

meeDamian, 2019-02-07:
Remove copies of files from the repo. Instead extract them from downloaded Alpine release in `make_img.sh` script.

AnotherDroog, 2019-02-13:
Many files will be generated by the programs on first start

It would be nice to avoid downloading the entire alpine release. If we need to fetch files that can’t be generated it might be better to download them from alpine web git

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

meeDamian, 2019-02-15:
Since it'll all happen on Travis, it's better to issue less requests, but cryptographically verify them.  As there's no point in optimizing for bandwidth/speed when it happens on an automated external pipeline ;).

AnotherDroog, 2019-02-15:
Gotcha, I misunderstood initially, thinking you meant remove from image

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

meeDamian, 2019-02-15:
Nope, just our repo :).

<small>Sent with Google Pixel 3, Android version 9 patch February 2019, using Sinet 5GHz WiFi and Google Chrome Beta version 73.0.3683.37</small>

AnotherDroog, 2019-04-22:
`Sent with Google Pixel 3, Android version 9 patch February 2019, using Sinet 5GHz WiFi and Google Chrome Beta version 73.0.3683.37`
🤣

Major cleanup coming in v0.5!

nolim1t, 2019-10-07:
I think this is still valid. Probably can start removing stuff going forward from the most recent release

PR #242 (2019-10-08): Cleanup and documentation
Resolves #151 and also resolves #241 

## Tasks TODO

- [x] Proofread documentation
- [x] Do a test run without any flags enabled
- [x] Do a test run with ALL flags enabled
closed 2019-10-08 by nolim1t