lncm/pi-factory #158: Can't restart or shutdown the box if it doesn't have internet access

nolim1t, 2019-02-10:
Can't restart / shutdown the box if it doesn't have internet access.

An edge case, but it may happen.

meeDamian, 2019-02-11:
😂

NoahO, 2019-02-25:
Can we make a list / test things that are not power-safe?

Generally, any database/filesystem should maintain a transaction log and thus be able to recover 100% from power off with no issues.. And unless it is hosted on a battery-backed RAID controller, the default is to wait until writes finish before moving on. 

We need to note if their are edge cases with lightning and how to address them. We don’t want critical data loss scenarios.

nolim1t, 2019-02-27:
This is one of them. Other than the usual lightning stuff with restoring channels (this is an LND issue, and affects more than just our  install).

I'm going to look at c-lightning as you can back up state a lot easier. Other than not having autopilot, it seems to be a lot more robust and production ready. And it can handle non-TOR and tor connections better.

AnotherDroog, 2019-03-05:
Test if fixed in master

AnotherDroog, 2019-04-22:
Box will continue to block shutdown / restart while critical parts of installation are running.

We stop blocking keyboard input and shutdown / restart behaviour and allow logins local console once `lncm-post` has run. Please re-open if this behaviour needs adjusting

commit 8edd17c0 (2019-04-17): lncm-post: Remove lncm-online dependency; fixes #158
closed 2019-04-22 by AnotherDroog