lncm/pi-factory #159: Add in IPFS

nolim1t, 2019-02-11:
Add in IPFS container into the docker compose.

This falls under the same philosophy of self-sovereignity, and where the more boxes that are deployed the more they should support each other. 

We won't be pinning any files yet, but would like to be able to have our website hosted on ipfs so that its constantly available.

AnotherDroog, 2019-02-13:
I’ll be using IPFS to distribute the blockchain snapshots we publish

Letting boxes then pin it and help sharing it would be in line with our aim of strengthening the network

Those who do not want to participate in sharing (possibly because of high cost volume metered internet or extremely low bandwidth) can turn it off in the dashboard

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nolim1t, 2019-10-07:
Lets hold on this for now

closed 2019-10-07 by nolim1t