lncm/pi-factory #164: Discussion: `apk` as deployment pipeline

AnotherDroog, 2019-02-13:
Let’s figure out if it makes sense to bundle our tools as a `apk` package for alpine. 

What are the pros and cons?

`apk add mana` would provide a convenient way to deploy

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

NoahO, 2019-02-14:
Cons: 
Another development pipeline
We already have Docker
If our tools have dependencies, they get installed onto the host OS
Any dependencies that get installed increase the security footprint for the whole system, not just one docker image.. 
If dependencies have vulnerabilities, they have to be updated too

meeDamian, 2019-02-15:
I generally agree with you @NoahO. Especially if we can do `build` stage and `final` stage on Docker.

closed 2019-03-25 by AnotherDroog