lncm/pi-factory #169: Create lncm/docker-mana repo for our stuff

meeDamian, 2019-02-15:
Let's create a `mana` image and start moving all our tools there.

AnotherDroog, 2019-02-21:
What do you mean by image? A container?

meeDamian, 2019-02-22:
Yes.

AnotherDroog, 2019-02-22:
I’m not sure managing the host makes sense from inside a container

meeDamian, 2019-02-22:
Sure, neither am I.

I was looking into `--privileged` flag the other day, perhaps it can be useful?

https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities

Now that I think about it, I think there are two aspects to this issue:

1. HW, OS and general cluster management
	1. managing docker
	1. managing storage
	1. managing updates
1. Purely app-level management, ex:
	1. `curl` to ddns
	1. Some app-specific crons
	1. `upnp` setting ports, although we might need to be smart about it…
	1. reconnecting `lnd` to its peers
	1. sending us diagnostic info (if enabled)

Putting the second one into a separate container might be beneficial, and allow for our stack to be runnable on other HW/OSs eventually (old laptop, or whatever).

AnotherDroog, 2019-03-18:
Created https://github.com/lncm/noma

closed 2019-03-18 by AnotherDroog