lncm/pi-factory #176: Consider switching away from Travis

meeDamian, 2019-02-22:
Super low priority, but I think it's good to keep a trace of it.

Last month Travis got acquired by a company known to buy companies, and then stop investing in them and just suck off their pre-existing revenue streams until they die.

SRCs:
https://news.ycombinator.com/item?id=18978251
https://news.ycombinator.com/item?id=19218036

nolim1t, 2019-02-27:
Maybe Gitlab CI? 

It's more decentralized as you can even host your own builder.

meeDamian, 2019-02-27:
We do CI to have auditability.  All auditability is lost if we control the builder.

AnotherDroog, 2019-03-04:
If we really do want to switch, or try an alternative, I would really like to check out https://sourcehut.org