lncm/pi-factory #177: Add repository wide linting

AnotherDroog, 2019-02-23:
I've fixed most linting errors and it would be nice to keep the repository clean going forward. 

### Let's integrate either of these services into our Github:

- https://codeclimate.com/quality/
- https://www.codacy.com/
- https://www.codefactor.io

### For your dev environment and Travis CI
- https://www.shellcheck.net
- `brew install shellcheck`

nolim1t, 2019-02-24:
good idea going forward. Maybe as a start add some [Contributing Guidelines](https://help.github.com/en/articles/setting-guidelines-for-repository-contributors) for this project

nolim1t, 2019-02-24:
@AnotherDroog - thoughts?

nolim1t, 2019-02-24:
Code Climate looks good, and its free for open source projects

NoahO, 2019-02-25:
Have you got any recommendations for linting plugins for local work too? 

For example I am very impressed by the VSCode plugin library, they have Dockerfile support and others. Should we start a wiki for recommendations setting up coding environment?

closed 2019-03-07 by AnotherDroog