lncm/pi-factory #178: Test various power supplies

meeDamian, 2019-02-24:
Buy different ones, and try to abuse them.  See which ones handle apocalypse the best.

AnotherDroog, 2019-02-24:
- Official Raspberry PSU (5.1V 3A) is good

The one 1.5A one bought at Computer Plaza for 79 baht is *not* good enough, undervoltage with 3 USB sticks attached.

- Bought one at 7-Eleven, 200 baht, 2A, 5V, works okay

meeDamian, 2019-02-24:
I was primarily referring to what @NoahO mentioned during our meeting in Cube7: Test how these power supplies would handle abuse in a form of power spikes etc.

NoahO, 2019-02-24:
Perhaps we can research electrical testing certifications and only use chargers approved by certain schemes? Setting up an electronics testing lab is bit out of scope for us..

meeDamian, 2019-02-24:
Right, that's much faster, cheaper and better 😅.

AnotherDroog, 2019-02-24:
Not sure about certifications we can rely on, especially with all the knock-offs and garbage around. 

We can do our own tests:

- test voltage is >4.8V (undervoltage cutoff)
- check there is 2A of current available

NoahO, 2019-02-24:
My point was that a lot of these chargers are cheaply made, and might go open circuit (eg 220V all the way to the USB port) if they fail/get exposed to a surge, etc and blow up everything connected.

But my real point was that we can't rely on connected USB drives for funds backup - it wasn't about hardware reliability. Even good chargers can die.

NoahO, 2019-02-25:
Maybe another thing we can do, is expose to the user when the power throttling is being hit on the pi, either by email notification or dashboard notification. 

You could also argue that the Pi does a really good job of throttling, and it shouldn’t be that much slower if they use a lower amperage PSU..

AnotherDroog, 2019-03-04:
Yes great idea, undervoltage detection will need #78 to work