lncm/pi-factory #181: wifi isn't working again at latest master

nolim1t, 2019-03-05:
Wifi isn't working again at latest master.  (no LNCM-Box AP or internet connection if I pre-loaded a WPA supplicant file)

To get it working, I restored to commit https://github.com/lncm/pi-factory/commit/6a8f7f071c0fc5dcbc24161a5f9e1077d7593a46

nolim1t, 2019-03-05:
it appears iotwifi doesn't get loaded

nolim1t, 2019-03-06:
[image]

Screenshot

AnotherDroog, 2019-03-07:
Please test latest release too

nolim1t, 2019-03-07:
Still not fixed at commit https://github.com/lncm/pi-factory/commit/b22b899332bc2f1518b2ff702aafec8e15528ad6

AnotherDroog, 2019-03-07:
how does this manifest?

nolim1t, 2019-03-07:
Removing all the files in lncm-workdir fixes the issue

closed 2019-03-07 by nolim1t