lncm/pi-factory #187: USB Stick formatting doesn't work

NoahO, 2019-03-09:
After leaving a text file "DESTROY_ALL_DATA_ON_THIS_DEVICE.txt" on each USB stick, it didn't format them or use them, and instead wrote to the SD card. 

Also related: https://github.com/lncm/pi-factory/issues/134

AnotherDroog, 2019-03-09:
Possibly related to disk labels bug #174

nolim1t, 2019-03-11:
it formats if the disk is ext4, however this does affect devices which aren't ext4

AnotherDroog, 2019-03-12:
I've got three new sticks that I can test with.

@NoahO if possible, please describe the sticks in more detail, ideally post the partition layout, e.g. from `parted`

AnotherDroog, 2019-03-16:
Found the culprit:

UUIDs change after formatting and the installation script did not account for that!