lncm/pi-factory #198: Configure docker so that it doesn't use up all the space on the SD card

nolim1t, 2019-03-27:
# Issue

After a while the SD card gets used up with logfiles

# Solution

Set a configuration value to stop the docker log files from taking up all the space, or disabling log files if it is possible as we already have logging (is this possible @NoahO ? )

# Where its an issue

On the food4thought box and my own box,

nolim1t, 2019-03-27:
Another temporary fix which stops logs completely (so you have to use individual app logs)

```
echo "" > $(docker inspect --format='{{.LogPath}}' compose_btcbox_1)
echo "" > $(docker inspect --format='{{.LogPath}}' compose_btcbox_1)
echo "" > $(docker inspect --format='{{.LogPath}}' compose_invoicerbox_1)
```

AnotherDroog, 2019-03-27:
Cool that's a good idea as a temporary workaround

AnotherDroog, 2019-03-27:
What if it's not the logs?

nolim1t, 2019-03-27:
it seems the main culprit is whats in containers.

nolim1t, 2019-03-27:
btw that above workaround doesn't stop the box logs, it actually just wipes it temporarily. 

So we don't have to restart docker-compose

We just iterate through all the containers (which I think we should include nginx and iotwifi), and pass

```
echo "" > $(docker inspect --format='{{.LogPath}}' containername)
```

and it will just reset the containers log files

AnotherDroog, 2019-03-27:
This ties in to #76

nolim1t, 2019-03-27:
Created issue within noma project to make this a utility for now, as configuring docker and reproducing this issue takes some time.

noaho, 2019-08-14:
I also doubt this is the logs. To clean up past docker images, builds, etc, use docker system prune command.

nolim1t, 2019-08-14:
Perhaps  we could put ```docker system prune --force``` on a crontab ?

closed 2019-10-07 by nolim1t