lncm/pi-factory #209: Update invoicer to 0.3.0

AnotherDroog, 2019-04-22:
Our `docker-compose.yml` is still on 0.1.1!

nolim1t, 2019-04-22:
Possible but we need to also setup a tool for generating the config. I believe we have the toml library

AnotherDroog, 2019-04-22:
Ok so this will be a more involved that expected. 

I pulled the toml dependency since that was part of `lncm-post` and it was not used anywhere. 

Best to simply add it to requirements.txt in Noma!

nolim1t, 2019-04-25:
Yes that's why we haven't used the latest version yet.

We also need to integration test it. 

I think first priority is to get a stable master OR release candidate that can be put in master.

AnotherDroog, 2019-04-27:
Ok, v0.5 is almost ready

nolim1t, 2019-05-23:
Starting to work on this as of https://github.com/lncm/pi-factory/pull/218

nolim1t, 2019-05-28:
Superseceded by #223

closed 2019-05-28 by nolim1t