lncm/pi-factory #229: Fastsync "You need to rebuild"

AnotherDroog, 2019-06-08:
Keep getting this after extracting `utxo-snapshot-bitcoin-mainnet-565305.tar` 
* checksums match `openssl dgst -sha256`
* `SHA256(utxo-snapshot-bitcoin-mainnet-565305.tar)= 8e18176138be351707aee95f349dd1debc714cc2cc4f0c76d6a7380988bf0d22`
* fresh `bitcoind` directory
* fresh `bitcoind.conf`

```
2019-06-08T10:30:05Z Bitcoin Core version v0.17.1.0-gef70f9b52b851c7997a9f1a0834714e3eebc1fd8 (release build)
2019-06-08T10:30:05Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-06-08T10:30:05Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-06-08T10:30:05Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-06-08T10:30:05Z Using the 'sse4(1way),sse41(4way)' SHA256 implementation
2019-06-08T10:30:05Z Using RdRand as an additional entropy source
2019-06-08T10:30:05Z Default data directory /root/.bitcoin
2019-06-08T10:30:05Z Using data directory /root/.bitcoin
2019-06-08T10:30:05Z Using config file /root/.bitcoin/bitcoin.conf
2019-06-08T10:30:05Z Using at most 125 automatic connections (1048576 file descriptors available)
2019-06-08T10:30:06Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-06-08T10:30:06Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-06-08T10:30:06Z Using 2 threads for script verification
2019-06-08T10:30:06Z scheduler thread start
2019-06-08T10:30:06Z HTTP: creating work queue of depth 16
2019-06-08T10:30:06Z No rpcpassword set - using random cookie authentication.
2019-06-08T10:30:06Z Generated RPC authentication cookie /root/.bitcoin/.cookie
2019-06-08T10:30:06Z HTTP: starting 4 worker threads
2019-06-08T10:30:06Z Using wallet directory /root/.bitcoin
2019-06-08T10:30:06Z init message: Verifying wallet(s)...
2019-06-08T10:30:06Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2019-06-08T10:30:06Z Using wallet wallet.dat
2019-06-08T10:30:06Z BerkeleyEnvironment::Open: LogDir=/root/.bitcoin/database ErrorFile=/root/.bitcoin/db.log
2019-06-08T10:30:06Z Cache configuration:
2019-06-08T10:30:06Z * Using 2.0MiB for block index database
2019-06-08T10:30:06Z * Using 8.0MiB for chain state database
2019-06-08T10:30:06Z * Using 440.0MiB for in-memory UTXO set (plus up to 286.1MiB of unused mempool space)
2019-06-08T10:30:06Z init message: Loading block index...
2019-06-08T10:30:06Z Opening LevelDB in /root/.bitcoin/blocks/index
2019-06-08T10:30:06Z Opened LevelDB successfully
2019-06-08T10:30:06Z Using obfuscation key for /root/.bitcoin/blocks/index: 0000000000000000
2019-06-08T10:30:12Z LoadBlockIndexDB: last block file = 1549
2019-06-08T10:30:12Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=66, size=59337972, heights=565529...565594, time=2019-03-03...2019-03-04)
2019-06-08T10:30:12Z Checking all blk files are present...
2019-06-08T10:30:12Z LoadBlockIndexDB(): Block files have previously been pruned
2019-06-08T10:30:12Z : You need to rebuild the database using -reindex to go back to unpruned mode.  This will redownload the entire blockchain.
Please restart with -reindex or -reindex-chainstate to recover.
2019-06-08T10:30:12Z Aborted block database rebuild. Exiting.
2019-06-08T10:30:12Z Shutdown: In progress...
2019-06-08T10:30:12Z scheduler thread interrupt
2019-06-08T10:30:12Z Shutdown: done
```

AnotherDroog, 2019-06-08:
Hashes here: https://github.com/btcpayserver/btcpayserver-docker/blob/master/contrib/FastSync/utxo-sets

nolim1t, 2019-06-08:
Just a note that fastsync is not in master, so this really belongs in the noma project which deals with orchestration now

AnotherDroog, 2019-06-13:
Found the issue: bitcoin.conf file had the wrong name 😅

closed 2019-06-13 by AnotherDroog