lncm/pi-factory #236: OS Only version of pi-factory

nolim1t, 2019-10-03:
Use the latest version of Alpine. 

Remove all the bitcoind / lnd node cruft from this and simplify it. 

Have the image script build just an alpine OS image which can easily install the ```noma``` package OR use for some other purpose.

## Requirements

- [ ] Support for ```aarch64```. The raspberry pi 3 is more than capable of handling this
- [ ] Keep support for ```armhf``` for the zero

meeDamian, 2019-10-06:
I would **love** to see this!

Simple way to bootstrap a minimal, RBP-bootable Alpine into an SD card!

😍

nolim1t, 2019-10-06:
> I would **love** to see this!
> 
> Simple way to bootstrap a minimal, RBP-bootable Alpine into an SD card!
> 
> 😍

see #238 

Testers welcome!

nolim1t, 2019-10-07:
On the tip of #238 we have a working pi-zero and 64 bit pi images on alpine with full ext4 persistance!

PR #238 (2019-10-07): (TESTING) Pull Request for OS Only pi-factory
Pull request for OS Only pi-factory (When merged this should close #236 )

## Contains
- [x]  Latest version alpine
- [x]  Modified wpa_supplicant script to detect for wpa_supplicant.conf file in FAT partition
- [x]  Latest version alpine cache
- [x] Pull request #237 
- [x] Github Action

## TODO

- [x] Test out Raspberry PI 3/3b
- [x] Create Raspberry Pi Zero alternate config
- [x] Test Raspberry Pi Zero alternate config

## Known Issues

- [x] Install fails if no internet, but this is handled. How to handle this in a userfriendly manner?

## Tes [... cut]
closed 2019-10-07 by nolim1t